nealcrook/multicomp6809 #2: 5 volt power to keyboard could exceed maximum input voltage on Cyclone II

Ian-May, 2015-10-03:
Hi Neal, I've been having a discussion with James Moxham about using 5V to power the keyboard instead of 3.3V. I have a KVM that wouldn't work on 3.3V so I hooked it to 5V and fitted two 3.3V Zener diodes to ground to limit the maximum voltage. I took some measurements without the Zener diodes and found there was 4.88V at the 10k pull-up resistors. I was expecting there to be diodes on the FPGA inputs to the 3.3V supply but it doesn't look like that is the case. According to the Cyclone II data sheet the absolute maximum input voltage is 4.6V and the maximum for 100% of the time is 4.0V. Have you fitted anything to limit the voltage? If you haven't you probably are risking your FPGA. BTW the limits on Cyclone IVs are 4.2V and 3.6V. Cheers, Ian.

nealcrook, 2015-10-03:
Hi Ian, thanks for your comments. To review, I have the kbd power wired to 5V and the pullups wired to 3V3. I expected that the kbd would present an open-drain on CLK and DATA -- in other words, it would actively drive them low but leave a high-impedance state on them when they need to go high. I expected that the pullups necessary to establish a logic-1 were the responsibility of the driving end (the computer motherboard or, in this case, the FPGA board).
After reading your comments, I measured the "idle" voltage on the SCLK/SDATA ends of my pullups and confirm your observation that they are closer to 5V than 3V3. This suggests either that the driver in the keyboard is not presenting a high-impedance load or that the keyboard has its own pullup, to its power rail (5V). If the latter is the case, there should not also be a pullup at the FPGA board end.
Your solution of a 3V3 zener to ground on each of SCLK, SDATA sounds like good idea. I will put some of these on my shopping list. I'm also inclined to take the keyboard (DELL) apart to see whether I can identify pullups on these lines.
regards, Neal.

nealcrook, 2015-10-03:
I updated 
https://github.com/nealcrook/multicomp6809/wiki/Debugging
to highligt this issue and add your suggestion of using zener diodes as clamps on these signals.

Ian-May, 2015-10-04:
Hi Neal, I took some measurements with 6 different PS2 keyboards, 2 USB keyboards with PS2 adapters and 2 ATEN PS2 powered KVMs. All of them pulled the clock and data lines to +5V with no load. I then measured the short circuit current to ground on the clock and data lines and got values from 1.1 mA (PS2 keyboard) to 4 mA (4 port KVM). These suggest pull up resistors of ~4.3k and ~1.2k. Whilst a 3.3V Zener would seem the best choice low voltage Zeners don't have a very sharp turn on characteristic so the high voltage is only 2.4V with the 4 mA KVM and 2.2V with the 1.1 mA keyboard (this is with the 10K pull ups to 3.3V installed). The high voltage for the FPGA is 1.7V minimum but I'm guessing the keyboard might have a TTL minimum high of 2.0V so 2.2V may be a little too close for comfort. I don't have any 3.6V Zeners to try but I think they may be the best choice so perhaps buy some of them. I did try 4.3V Zeners and the high voltages were 3.2V and 3.0V but protecting a 3.3V input with a 4.3V Zener doesn't seem right! I've also tried 1N4148 diodes to the 3.3V supply line and got 3.9V with the 4mA KVM. A BAT86 Schottky diode to 3.3V reached 3.5V. A Zener will also clamp a negative going voltage so that is why I think it is the best choice.
Cheers, Ian.

Ian-May, 2015-10-04:
I need to learn how to use this - I didn't mean to close the issue - sorry. Ian.

nealcrook, 2016-09-28:
BTW, the Jame's Cyclone II rev C PCB incorporates these diodes
https://www.retrobrewcomputers.org/doku.php?id=boards:sbc:multicomp:cycloneii-c:start
Anyone in Europe who wants one of these boards can get one from me at cost (~£2+postage). There's also a rev D PCB that adds additional level conversion/clamping options in this area, but I have not found that necessary.