nealcrook/multicomp6809 #3: Microsoft basic/Nitros9

bnobel, 2015-11-16:
Hi Neal,  just compiled and succesfully have a full 1mb version of your build working on my DE0 Nano.  Question I have is for specifically Basic & Nitros9.  Do I make an image file on the SD for Grants Microsoft Basic?  And also for Nitros9 I can definitely help with the port across for Level 2, as I helped create it in the first place.  Kernel files are the only mods needed for your MMU and Timer IO.

Bill Nobel

nealcrook, 2015-11-17:
Hi Bill,
do you have camelforth running (from Rom)? Camelforth has words BASIC BUGGY FLEX CUBIX NITROS9 to start the respective software by loading it from SD. The script bin/build_sd_image puts the images into specific offsets on the sd card. You can comment out any images you do not care about. So if you just reference the basic and the nitros disk image those are (of course) the only thjngs you will be able to start.
Are you running mercurial? I can send you a patch set with the mc09 additions.
Neal

bnobel, 2015-11-17:
Hey Neal,  Yes I have Camelforth running.  I was reading the documentation in the Wiki and noted the script for the FLEX disk image setup. I did’t see the full script in the repo had the missing pieces for Basic, Nitros9 and Buggy.  I am building a image now.

For the Nitros9 images are they currently limited to the standard floppy size for now?  I am starting to look at porting Nitros9 Level 2 to the Multicomp, but still wrapping my head around your MMU scheme and what I need to change in the Kernel.

Bill Nobel

bnobel, 2016-01-25:
Hey Neal,  I think I have a way to fix the PS/2 keyboard to get all the keys working accordingly.  I am not a pro at VHDL, but I think I have the idea of what needs to be fixed.  I need to add an extra conversion table for the extended codes, that the make codes start with $e0.  This should’t be too much of an impact on the hardware side.  I have also been thinking of stripping the FNtoggledKeysSig conversion to bring the FN keys back to the ASCII code table.

Bill Nobel