nealcrook/multicomp6809 #4: Microcomputer4 Build failed

douggilliland, 2017-01-25:
I did a fresh clone and tried building Microcomputer4. I got the following error:

Error (10344): VHDL expression error at Microcomputer.vhd(405): expression has 3 elements, but must have 7 elements
Error (10346): VHDL error at Microcomputer.vhd(394): formal port or parameter "cpuAddr" must have actual or default value

Line 405:
            cpuAddr => cpuAddress(15 downto 13),

MEM_MAPPER2.chd has:

        cpuAddr : in std_logic_vector(15 downto 9);

nealcrook, 2017-01-25:
Hi Doug,
Use microcomputerPCB instead. Sorry , I keep meaning to fix up microcomputer4 but really it has been superceded by microcomputerPCB. The latter is compatible but has more features.
Neal

douggilliland, 2017-01-26:
Thanks! That worked great. I have been trying to get the 6809 multicomp working for a long time and it was buggy. I could get the 2K internal RAM working but that doesn't make too satisfying a system. I was able to use yours and replace it with the BASIC ROM from Grant's and it worked good.
Now I need to figure out how the mapper works in yours to make it work with [the card I designed](http://land-boards.com/blwiki/index.php?title=EP2C5-DB).

nealcrook, 2017-01-27:
Hi Doug,glad that worked for you. There are a lot of improvements and timing rework compared with Grant's original. Ms basic gets boring (again) after 5 mins when you realise that there is no way to save/load. If you use my rom image you can put my pre - build disc image onto Sd card and run the same ms basic image but loaded from Sd.  You can also run flex, nitros9, etc etc.
Neal

closed 2017-05-17 by nealcrook