nealcrook/multicomp6809 #5: Source code for Nitros09 port

znac049, 2017-04-19:
Hi Neal, 
is the source to your Nitros port available anywhere? I'd be really interested in taking a look as I'm building a few Multicomps (as soon as the PCBs arrive!). I'm also doing a non-FPGA version of the Multicomp based on the 6309 (with MMU in CPLD and using CF instead of SD) and would prefer not to have to reinvent too many wheels. Hope you don't mind me asking.
Thanks,
-Bob

znac049, 2017-04-19:
Doh, I found your code in the official Nitros repo on sourceforge. Safe to close this now :)

closed 2017-04-19 by znac049

nealcrook, 2017-04-19:
Hi Bob,
the source code for the level1 port is in the sourceforge repository  (target name is mc09)and it should build and run.
The source code for the level2 port is part - present  in the official repository (target name is mc09l2). I am working on getting the remaining files into the repository.. should be done in the next 4 weeks. I could email pre release files sooner. Ping me when you are ready for them.
Good luck with the build. Feel free to email me if you get stuck. Your 6309 project sounds interesting. Will you be using my mmu design (fastest route to getting nitros9l2)?
Neal

nealcrook, 2017-04-19:
BTW, I'd be interested in reviewing/critiqueing your 6309 design.. especially if you're using a 6309E (I have had a couple of these parts in my junk box for about 20 years..)
Neal.

znac049, 2017-04-19:
Hi Neal,

Thanks for getting back to me – I’ll let you know if I need the l2 files in advance of you putting them up on sourceforge. I ordered a batch of V110 PCBs from DirtyPCBs a couple of weeks ago, so it might be a few more weeks before they arrive.

I did a fair bit with the 6809 back in the day and it’s still my favourite 8-bit processor. The 6309 looks like fun, too and like you, I have a few in a box somewhere that are just crying out to be put into service. I have 63C09 and 63C09E cpus and hopefully have a design that will support both, CPLD permitting. I plan to use your MMU, at least initially, but have an idea for an extension; I think it would be fun for certain IO locations only to be accessible to the OS (supervisor mode) and I think with suitable monitoring of the BA and BS signals, it should be possible to do that, so an interrupt vector fetch will enable access to the protected area and the last thing the interrupt handler does before returning is to disable the protected area. I’m going for CompactFlash for no good reason other than it feels like a better fit than SD, (and easier to interface) in the context of a 1980’s system. I hacked a version of exec09 with a fairly complete 6309 instruction set and an emulated CF interface and can boot a test program from an OS9 formatted CF.

I’d be happy to send you the schematic for the 6309 computer when it’s a bit firmer and would welcome feedback. I’ve done a bit of prototyping but really don’t get on with breadboard so will commit to a PCB quite early on – I’d be happy to send you one if you’re interested. MMU and glue logic will be in CPLD (something like Altera EPM7128 which comes in 84pin LCC) but the ram/rom/cpu/uart/CF will be original parts.

Cheers,
-Bob

P.S. my direct email is bob@wookey.org.uk