nealcrook/multicomp6809 #6: How to test full 1M external RAM

jbforrer, 2017-05-08:
Hi Neal, 

First, thanks for your efforts. Great job and a fun project.

Got the Multicomp6809 working nicely with 1M external RAM, though not sure all RAM is detected.

I'm still exploring making SD images and am using a pre-built image from RBC repository; there are two Jan2016 and April2017. I've built the FPGA image from your latest GIT repository. For some unknown reason the April2017 SD image fail to boot NITROS9 in my setup ... it comes up as far as the date prompt, then dies. However, the Jan2016 SD image works fine; BUGGY, CUBIX, FLEX, NITROS9 & FUZIX all loads and runs fine.

Q: FUZIX reports 512K RAM although 1M is installed. Any suggestions how one can test/verify the full compliment of RAM ?

Incidentally, found a broken solder joint on my module affecting PIN135 that caused RAM failures (A4 was stuck). I wrote a diagnostic VHDL test to cycle all address lines and noted the FPGA was OK but PIN135 was floating. Repair made with fine wire and all RAM tests 0800 - CFFF passed.

Best Regards.
Johan Forrer, 
Eugene Oregon.

nealcrook, 2017-05-09:
Hi Johan,

thanks for your kind words and congratulations on getting your board up and running. 

The FUZIX code is (currently) hard-wired for a memory size of 512Kbtytes. I will add memory sizing to my "to do" list.

The two SD images should only differ in their NITROS9 disks. The old one contains the NITROS9 "Level 1" port, which can only access 64Kb of memory. The new one (April 2017) contains the NITROS9 "Level 2" port, which can access upto 1Mb of memory -- though currently it is (like FUZIX) hard-wired to use 512Kb (adding memory sizing is already on my "to do" list and should be done in the next few weeks).

Your system is hanging during the NITROS9 Level2 boot. There is only one piece of hardware used there that is not used for any other software, the "FRT mode" of the memory mapper. I *think* that is part of the RTL release on RBC. To be sure, please check the file mem_mapper_mk2.vhd (I think that's the name) and look for the text "FRT". If it is missing, you will need to get the latest code from my github repository and rebuild the FPGA (..and I will have to do a new release).

If that's not the problem, the next thing I suggest is removing the 2nd memory chip (it isn't being used and perhaps it's causing a short, though the memory test and the fact that FUZIX is working seem to make that unlikely)

Was your broken solder joint on the Cyclone FPGA board or the V110 PCB? I had a solder short on one of my Cyclone FPGA boards.

Try these ideas and let me know if it solved your problem. I'll try to think of some other things to check.

Neal.

jbforrer, 2017-05-09:
Thanks for the quick response Neal. Good to see you are still interested in this project -- it has been very educational and challenging to me. 

Understand the 512K limits and looking forward for your future updates, however works great as is for me. I swapped out the set of RAM chips and now find the April 2017 SD image and current VHDL from your GIT repository now all working as expected. Perhaps one of the RAM chips was not behaving well at 3.3V?

Incidentally, built my little system on perf board, mostly wire wrap and some soldering --- much like your prototype. Included the RTC part as well which comes in handy with NITROS9. The VGA and keyboard does work fine, but prefer to use the console redirect to a console session on my Linux host.  

FUZIX and NITROS9 are amazing, will have to explore your port to the multicomp6909.

Thanks so much for your support.

Regards.
Johan.

closed 2017-05-09 by jbforrer

nealcrook, 2017-06-09:
Hi Johan,

The latest version of NitrOS-9 Level 2 for multicomp6909 will auto-detect between 512KByte and 1MByte memory sizes. When the system has booted, use the MFREE or MMAP commands to "prove" how much memory you have.

You can rebuild from source (on Sourceforge) or download the pre-build image (the _dxl image) from  https://nitros9.sourceforge.io/snapshot/). You will need to regenerate the SD card image or use a script like my bin/replace_nitros_disk0 to replace the NitrOS-9 image. Let me know if you get stuck doing this.

I'll make a similar fix to FUZIX next time I work on it, hopefully in a month or two.

Neal.

jbforrer, 2017-06-09:
Thanks much Neal, will check it out shortly.
Johan.