nealcrook/multicomp6809 #8: Bringing up multicomp stuff on RetroShield6809

erturkk, 2019-08-23:
Hi Neal,

Would it be possible if I reuse some of your OS files on my open-source hw project?  I didn't see a license file, so wanted to check with you.

I started an Arduino project called "RetroShield for Arduino Mega":

www.8bitforce.com/projects/retroshield
www.gitlab.com/8bitforce/retroshield-hw
www.gitlab.com/8bitforce/retroshield6809

Basically I hooked up a 6809 to Arduino Mega and emulate all system hw in Arduino except the real processor executes code. Similar to your fpga design. I ported the Buggy monitor code that I modified for my older Simon6809 project. Next step is to port the Flex09. Ultimate goal is to use SDCard as HDD and a VGA/Keyboard shield for Arduino for a stand-alone computer :)

If you like to bring up the OS together, I'll be up for that too.

Thank you,
Erturk

nealcrook, 2019-08-24:
Hi Erturk,
the stuff on my multicomp pages comes from a lot of different places so I could not specify a license. Also, some of it (like the Flex binaries) does not belong to me, it is just "abandonware".

Having said that, yes: you are welcome to use any of the files for your project. I had read about your "Retroshield" project on one of the forums, but it is only now that I realise that you also created "simon6809" which I read about long ago.

Ever since I read about  https://hackaday.io/project/159973-z80-mbc2-4ics-homemade-z80-computer I have thought about doing a 6809 version, so I am pleased that you are doing it.

FLEX should be straightforward to port, and I hope that the guide on my WIKI helps you through some of the steps. I do have 2 63B08E chips in the drawer so I will add this to my list of "things to build". Let me know if you get stuck on anything.

Neal.

erturkk, 2019-08-24:
Great, thanks Neal. Wiki will be super helpful. You put allot of work into it. What is your address? I will send you pcbs for those two 6309. Don't let them sit idle. Plus you can help me when I get stuck :)You can email me at erturkk at 8bitforce dot Com. Enjoy your weekend, Erturk