nealcrook/multicomp6809 #10: Issues with my  attempt to compile

okwatts, 2020-07-27:
Sorry if this is the wrong way to communicate but I am having trouble compiling the project as it gives an error saying that Error (10481): VHDL Use Clause error at SBCTextDisplayRGB.vhd(295): design library "work" does not contain primary unit "SansBoldRom".
 In addition I have been using the bare minimum internal Ram  version to debug. Using the pof in the zip file I can run this and use Camelforth but the SD light is on (no sd card holder present at the moment?) and the reset key has no effect it appears that line is permanently low. I have got the sera line working with a terminal but the external ram is flaky will not stay consistent. If I remove the fpga board from the v1.10 board the reset symptoms are the same. As you can tell I am a newbie at this FPGA stuff but have managed to get Grant's bare bare minimum versions for the 6502 and Z80 to work as well as the multicomp08_iram version. I have ordered another FPGA board to see if this one is wonky, just waiting for arrival. Thanks Larry

nealcrook, 2020-07-27:
Hi Larry,
the code supports 2 different fonts. One is stored in files Components/TERMINAL/CGABold* the other in Components/TERMINAL/SansBold*. In SBCTextDisplayRGB.vhd this line selects (at compile-time) which is used:

constant SANS_SERIF_FONT : integer := 1 -- 0 => use conventional CGA font, 1 => use san serif font

So, setting 0 should solve your problem.. but you should not be seeing this problem...

The top-level project/design should be "MicrocomputerPCB" - that is the latest. If you are using one of the other top-levels the problem is probably that the SansBold* files are missing from the MicrocomputerPCB.qsf file (you can add them from the GUI or by hand-editing the file).

The SD LED should not default on (with or without the SDcard fitted) and the reset key should work! It might be worth taking a look at the cyclone board schematic (on Grant's site) and working out what are the adjacent pins on either side of those two pins, then buzzing out to see if there's a short. 

The fact that you have successfully built the z80 and 6502 versions working shows that you have climbed the learning curve to a useful degree! There are lots of z80 systems around but I haven't heard of anyone building the 6502 variant; what software are you running on it?

For reference: are you building on Linux or Windows?

Neal.

okwatts, 2020-07-27:
Thanks for your prompt reply. I used the code from this GitHub and changed
for the minimum system and it compiled fine. I had been using the iram zip
from the retrobrew site and while the files were there the .qsf had not
been updated. My lack of knowledge using quartus is showing. I have been
using windows for this but will switch to Linux at some point. I have
continued weirdness as there are 2 LEDs lit (SD and the next one) at least
the one for the serial port behaves as it should. Reset still doesn't work.
I can't find any shorting in the immediate area of the pins. I tried with a
scope to look at the reset and when I could get it to trigger properly it
appeared to high and then drop and stay low. I have tools but the knowledge
is deficient it appears. Thanks again I will keep trying as I ordered 10
v110 boards from dirtypcbs so more cyclone2 boards will hopefully help me
get to the bottom of this.

nealcrook, 2020-07-27:
since you have a scope.. create a new project (clone the files of the existing project) keep the port list and delete all of the components. Code a counter and tie the counter outputs to pins assigned to LEDs, address lines etc. You can even redefine RESET pin as an output and drive that too. Then, with oyur scope you can check that all the pins are toggling cleanly and at the right frequency - any shorts will be obvious.
Neal.

okwatts, 2020-07-27:
Thanks for the suggestion, I will give it a try (will take an introduction to really using VHDL before I can report on this.)
Will update this when I have some results.

okwatts, 2020-07-30:
Hi Just an update to say that while I have yet to track down the issues with the original FPGA board, my second board arrived today. I installed that and the external memory checks out fine and the reset button works. The SD led and the one next to it (D4 on the FPGA board) still stay lit with the your version of  multicomp.  ( I recall that with Grant's only the SD led was lit). I checked that the SerA works but I am still awaiting the SD card holder to go further. At least I know the problem is on the original FPGA. I may try to get to the bottom of it or I may just try and return it for another. It seems it might be like a lottery as to the quality! Also I have yet to try the RTC (actually I can't yet) but the board I got has the pins facing the top side so I will have to desolder and reverse so I can install on the board. I will check it with an Arduino and then use jumper wires to test before desoldering.
Thanks again for the work and comments on the wiki. I look forward to fully using the software for the SD card. If I can I will have a go at the 6502 as my first micros were a KIM and SYM (which I still have) I used  DOS/65 (a CP/M alike from Richard Leary) on those using a home built floppy controller (still have the drives and media!) I will likely borrow from your work and the CP/M versions to help me get it to work with the SD as you have done.

okwatts, 2020-08-09:
Just an update to say I have received 3 FPGA boards and each of them has had a problem. My original board had 4 pins on the FPGA in the 141-144 area that did not connect to the pads on the circuit board through misalignment or poor solder. I managed to see this with a microscope camera and was able to correct them with a fat soldering tip and a sharp needle to align them. My second board seemed to work well until I tried the SD card. Turns out the sdCS line is at about 1.6 V and won't go lower so never get to select the SD. I will try and work around that. The last board seems to have sramAddress(2) stuck at ground haven't been able to determine why but again may ask for a replacement or try another pin. It seems pin76 on the FPGA is about the only free pin for output if you want to keep all other options available.

okwatts, 2020-12-18:
Hi Neal
I have been using and appreciating your 6809 additions and work to make a pretty complete ability to run most non graphical programs for this CPU. I have a question as I contemplate moving to implement a 6502 version. As one who has very little VHL experience I looked over your version and to my untrained eye, I would expect that apart from the clock modifications where you use hold to limit rate of execution a good deal of the added and improved code could translate to the 6502, ie GPIO,MMU and SDCARD. Do you have any comment on this or is my lack of understanding so complete that I should just stay away from it entirely.
Thanks
Larry

nealcrook, 2020-12-19:
Hi Larry,
I am glad you continue on your Great Learning Experience (I do prefer this to soldering nowadays!) As far as I know nobody has done any evolution of Grant's original 6502 design so I think  this is a great project. I am sure there is lots of 6502 software around in one form or another that could be leveraged (eg, Wozniak's sweet16 monitor).

I think you have correctly scoped the work required. I strongly recommend that you get up-and-running with the Modelsim simulator that is part of the Quartus package, because this provides the best method for debugging the clocking/control changes that you will need to make. Get started by getting the simulator running on the 6809 board -- there are some details on my wiki pages here. Next, try writing a tiny loop in 6809 code and put it in the ROM, and see that you can simulate it. Armed with this experience, you can rip out the 6809 and bolt in a 6502. Create a simple code loop in a ROM for the 6502 and you will have pretty much everything you need to debug the system.

Let me know if you get stuck or need more detailed tips. I am *not* a VHDL expert (I prefer and usually use Verilog) but for coding simple combinational and clocked logic, the existing top-level code should provide examples of all of the structures you need. If you understand enough digital electronics to know what you're trying to create, it should be straightforward (especially with the aid of the simulator) to code.

regards,

Neal.

okwatts, 2020-12-19:
Hi Neal
Thank you for your reply.  I really don't know either VHD, I ventured in
because of Grant's and your postings. I will do as you suggest and try to
understand using modelsim. I have had some success using what Grant
provided.  I have located 6502 code from Maik Merten to read Fat 16 on the
Multicomp (some issues but enough that I can load code). I have installed a
6502 monitor program from Daryl Rictor (6502.org and his own site
sbc.rictor.org). I can read and write sectors from the SD card thanks to
Maik's code and have gotten fig-forth and Carl Moser's Assm/ted running. I
really need to get an operating system going on it. I have a couple in mind
but the hard work is in front of me. One reason I would like to add your
modifications is to really use the GPIO and at some point the MMapper. I
want the Realtime clock module working in 6502 for date stamping. There is
a bit-banged IDE interface (through a IDE-CF unit) from a VIA that works
with an OS (symbiosys) for the SYM that I have working on my old SYM-1. I
have enough pieces to see where I would like to go but getting started and
pushing ahead will be a long process as it has been a long time since I
have programmed much myself. I can follow others and see and make
modifications in their code so gluing it together is more likely than
writing from scratch. Still as you have indicated it is a "Great Learning
Experience". I really appreciate all the work you did in getting all the
code working so well for the 6809 particularly as I consider what I want to
do for the 6502.
Thanks again for your work (and Documentation) that has inspired and
allowed me to revisit past computing eras ( I owned KIM, SYM, COCO
(especially OS-9), Atari ST, as well PC's over the years)
Larry

nealcrook, 2020-12-19:
Cool; sounds like you have made lots of progress with Grant's basic design. There's a really nice article on FAT32 here: https://www.pjrc.com/tech/8051/ide/fat32.html which makes basic accessing seem quite straightforward. I suggest porting the SDcard logic from my 6809 design because it has support for larger cards (the small card supported by Grant's original design are not so easy to find now). In general, the versions in my code-base have been cleaned up in various ways and should synthesise more cleanly and with a faster clock.

I think if you get modelsim going you will be able to inspect the 6502 running, and the 6809 running so you will have a good idea of how to control the 6502 in a new design based from my 6809 stuff. It looks as though you can use either ENABLE or RDY to stall the 6502. I don't want to spoil your fun, but if you get stuck feel free to ask for help.

okwatts, 2022-05-22:
HI Neal
After 2 years I have made little new progress on the 6502 front (at least
on the Multicomp). I have been on the 6502 forum (6502.org) and have
followed some of the people who have contributed there (especially plasmo
who is active on the retrobrew computers forum on many historical CPU's not
just 6502). I circled back to check out your Github and noted you have been
working away on the Nascom. I know little about that historical kit and see
that you even built a board to more fully represent it in newer form. I did
note that it looks like it was based on the Multicomp (ver 1.10c and FPGA)
with extra specifics like extra VGA etc. Does a version of this run on the
Multicomp you used for the 6809? I ask because I have extra boards and FPGA
and would be interested in exploring what you have done(, in a limited
manner of course.) The Z80 is a CPU I never did much with and while it had
a huge impact in the early years especially with CP/M, I managed only to
view from afar. In the years remaining it's always a conundrum whether to
try and stay current with the latest tech or retreat to retro (simpler?)
times even if I'm forced to learn some of the newer tech like CPLD/FPGA to
accomplish that.
Anyhow I have enjoyed your GitHub and Youtube postings and would like to
give Nascom a look, so let me know if it is possible( I don't think I am
ready to request a Nascom4 board at this time as it would likely remain
mostly empty.)
Larry

nealcrook, 2022-05-24:
Hi Larry,
"plasmo" is so prolific! I've enjoyed looking at lots of his 68xxx designs. One day, I will build one (I have a 68008, several 68000 and a 68020 in my junk box!)
My early "Nascom" prototypes used the Multicomp PCB & FPGA. For the full PCB I changed a lot of the FPGA pin assignments to make the PCB easier to route. 
If you are able to rebuild the (latest) FPGA code it should be straightforward to re-target it for the Multicomp PCB. It would be best if you connected push buttons for "reset" "warm reset" and "non-maskable interrupt". and LEDs for "HALT" and "TAPE".
You will end up with PS/2 keyboard, VGA screen (switchable between 48-column and 80-column), SDcard storage and a serial port. You will be able to run all NASCOM software, including PolyDOS and you will be able to run CP/M with a paged RAM disk. No modifications of my software should be needed.
If you get started and get stuck I will be able to drop you hints!
Neal.

okwatts, 2022-05-24:
Thanks for the reply I'll give  it a try and look at the pin assignments on
the schematic and in the pin planner.

okwatts, 2022-06-13:
Hi Neal I'm beginning to think this should be in the Nascom file. I have tried to get Nascom4 going on my multicomp and while I can see the SBD on the screen, the keyboard is not registering. If I boot with no SD card and use the warm reset I see the NAS-SYS3 but there is often a string of characters "gh..." and the PS/2 Keyboard doesn't generate any response. I did once after pressing on the FPGA board manage to actually get it to operate and could boot Polydos then it disappeared and never was seen again. It makes me suspect something is floating, but I have tried a bunch of things and nothing has worked. I include the QSF file which shows the pins I have assigned. Perhaps you could look it over and see if I have made a silly mistake. I know I am close but have not succeeded to make it reliable. I have used the exact setup of FPGA and Multicomp to run other versions like my 6502 software and I believe the hardware to be reliable. Thanks
[NASCOM4.zip](https://github.com/nealcrook/multicomp6809/files/8894805/NASCOM4.zip)
.

nealcrook, 2022-06-14:
I think your guess about a floating node is correct. There is some magic to choose between the ps/2 and nascom keyboard. In the latest nascom4.vhd (i revised it yesterday) look at lines 851-855 and identify the equivalent lines in your copy. Please try commenting out the IF so that you always get data from ioprd00.

okwatts, 2022-10-11:
Hi Neal
That seemed to fix it. Now to learn more about Nascom!
Thanks
Larry

nealcrook, 2022-11-08:
Cool! Glad you got it running successfully. You might like to join the Nascom group on groups.io
Neal.

nealcrook, 2023-01-14:
OK, I have retired the Microcomputer4 tree, leaving just MicrocomputerPCB. This should prevent any future confusion.